UCSolarCarTeam/Epsilon-Embedded-Software #40: CAN address/identifier issue- FilterIdHigh and FilterIdLow related

shilpan97, 2017-12-02:
Changing the CAN address/identifier does not send command to the controller. It drops all the identifiers that are not in the range. Bill and I believe it may have something to do with the FilterIdHigh and FilterIdLow for filtering the registers. Please investigate this issue as it critical for all embedded testing.
Thanks.

bill-luu, 2017-12-03:
Adding on to this, we were investigating this piece of code:
`````
void HAL_CAN_RxCpltCallback(CAN_HandleTypeDef* hcan)
{
    CanRxMsgTypeDef* msg = hcan->pRxMsg;
    HAL_GPIO_TogglePin(LED_RED_GPIO_Port, LED_RED_Pin);

    if (msg->StdId == LIGHTS_INPUT_STDID)
    {
        lightsInputs = msg->Data[0];
    }
    else if (msg->StdId == DRIVERS_INPUTS_STDID)
    {
        driversInputs[0] = msg->Data[0];
        driversInputs[1] = msg->Data[1];
        driversInputs[2] = msg->Data[2];
        driversInputs[3] = msg->Data[3];
    }
    else if (msg->StdId == BATTERY_STAT_STDID && msg->DLC == 4)
    {
        batteryStatus[0] = msg->Data[0];
        batteryStatus[1] = msg->Data[1];
        batteryStatus[2] = msg->Data[2];
        batteryStatus[3] = msg->Data[3];
    }

    __HAL_CAN_CLEAR_FLAG(hcan, CAN_FLAG_FMP0);
    HAL_CAN_Receive_IT(hcan, CAN_FIFO0);
}
`````
When we hard coded the BATTERY_STAT_STDID comparison to `0x703U` (the driver input value), messages would be accepted, but when we try to hard code the previous value to `0x6FBU` (The old battery status) or to `0x303U`(The new error status), it won't work.

Looking at this `main.c`:

````
static void MX_CAN2_UserInit(void)
{
    CAN_FilterConfTypeDef sFilterConfig;
    sFilterConfig.FilterNumber = 0; // Use first filter bank
    sFilterConfig.FilterMode = CAN_FILTERMODE_IDLIST; // Look for specific can messages
    // sFilterConfig.FilterMode = CAN_FILTERMODE_IDMASK;
    sFilterConfig.FilterScale = CAN_FILTERSCALE_32BIT;
    sFilterConfig.FilterIdHigh = LIGHTS_INPUT_STDID << 5; // Filter registers need to be shifted left 5 bits
    sFilterConfig.FilterIdLow = BATTERY_STAT_STDID << 5; // Filter registers need to be shifted left 5 bits
    // sFilterConfig.FilterIdHigh = 0; // Filter registers need to be shifted left 5 bits
    // sFilterConfig.FilterIdLow = 0; // Filter registers need to be shifted left 5 bits
    sFilterConfig.FilterMaskIdHigh = DRIVERS_INPUTS_STDID << 5; // Unused
    sFilterConfig.FilterMaskIdLow = 0; // Unused
 ....
}
````
I have a hunch that it's something to do with the filters, but we couldn't find much on them.

bill-luu, 2017-12-03:
Just kidding we have a wiki on can filters right here!
https://github.com/UCSolarCarTeam/Recruit-Resources/tree/master/Development-Guide/CAN-Filter-Guide
#itsallconnected

I'll take a look

bill-luu, 2017-12-03:
^ so that wiki didn't help but I managed to figure it out.

The line
`sFilterConfig.FilterMode = CAN_FILTERMODE_IDLIST;`
Sets the configure mode to look for specific messages, in `FilterId` and `FilterMaskId`. 
In 16 bit mode, we can have 4 specific messages to filter, stored in `FilterId.high`, `FilterId.low`, `FilterMaskId.high`, and `FilterMaskId.low`. 
However, this line: 
`sFilterConfig.FilterScale = CAN_FILTERSCALE_32BIT;`
Sets it to _32 bit_ mode! So we can only filter out 2 messages with sFilterConfig. Since we need to filter out 3 messages, and since the driver inputs on 0x703 are more than 16 bits long, I've added in another filter to look for battery errors. We can have up to 27 filters, so that makes life easier.

Further reading: Page 1080 onwards of the [STM manual](https://l.messenger.com/l.php?u=http%3A%2F%2Fwww.st.com%2Fcontent%2Fccc%2Fresource%2Ftechnical%2Fdocument%2Freference_manual%2F3d%2F6d%2F5a%2F66%2Fb4%2F99%2F40%2Fd4%2FDM00031020.pdf%2Ffiles%2FDM00031020.pdf%2Fjcr%3Acontent%2Ftranslations%2Fen.DM00031020.pdf&h=ATO_9tvuOLMVCtdpAeONcS0tDKj_dDtKJcIO0aWOLfQpp0dveHlrLJtam_ZpF1jMEpl8JqA2afoQIbvpBGqfkJaBf3RYe6fokd8LvijF5X35cz5dZmjiKp67gw8ZZCDfwsT-Jw).

closed 2017-12-03 by bill-luu