UCSolarCarTeam/Epsilon-Embedded-Software #66: Set Current Derivation

bill-luu, 2018-05-05:
CCS currently has fields for the setCurrent & setVelocity fields. However they are never being set, where are they supposed to come from?

enochtsang, 2018-05-06:
Those are from driver controls. It's the current and velocity we set from the driver controls.

bill-luu, 2018-05-08:
Between CAN & Serial, the Set Current is reported as a percentage. In internet reporting, the value is supposed to be in amps, how is this conversion done?

enochtsang, 2018-05-19:
I think it should always be in amps, I'm not actually sure of this atm, you should set it to what you see fit. It's basically a way to report driver input, don't worry too much about getting it right according to something, it's completley something we created.

closed 2018-05-22 by bill-luu