UCSolarCarTeam/Epsilon-Embedded-Software #131: Lights board not reporting lights status sometimes

bill-luu, 2019-07-22:
During the race, we noticed this behavior:

- Lights board would send and receive messages as expected
- At seemingly random times, the lights board would no longer send status updates, but the heartbeat still works and the lights themselves would still respond to the driver,

Reddragoneer1, 2019-08-08:
To clarify, the lights board would still send messages but sometimes the messages would only be heartbeat? 
The lights board still functioned as normal but did it display on the dashboard? 
Were we just not receiving it on the telemetry end?

bill-luu, 2019-08-11:
On the telemetry end, we would sometimes see it that the heartbeat message was working, but status updates would no longer send. 

Restarting the lights board by replugging in CAN would make it work again.